jbuehl/solaredge #149: Has solaredge switched to SSL?

hydrael, 2020-03-29:
I own a SE5000E Inverter and tried using `semonitor.py` in active Ethernet mode.
My inverter gets its IP address from semonitor and then tries to resolve `prodssl.solaredge.com`.
That basically happens in an endless loop and looks like this:

```
waiting for DNS message   
192.168.10.2:35345 --> message: 321 length: 39   
id: 363a 
flags: 0100
question
    name: prodssl.solaredge.com
    type: 0001
    class: 0001
192.168.10.2:35345 <-- message: 322 length: 76
id: 363a
flags: 8000
question
    name: prodssl.solaredge.com
    type: 0001
    class: 0001
answer
    name: prodssl.solaredge.com
    type: 0001
    class: 0001
    TTL: 86400
    resource: 192.168.10.1
waiting for DNS message
192.168.10.2:35345 --> message: 323 length: 39
id: c0c8
flags: 0100
question
    name: prodssl.solaredge.com
    type: 001c
    class: 0001
192.168.10.2:35345 <-- message: 324 length: 76
id: c0c8
flags: 8000
question
    name: prodssl.solaredge.com
    type: 001c
    class: 0001
answer
    name: prodssl.solaredge.com
    type: 001c
    class: 0001
    TTL: 86400
    resource: 192.168.10.1
```

Other than that nothing happens.
Could it be, that my inverter wants to talk SSL which is not supported by semonitor?

Side note: I have not extracted the encryption key yet, because my RS485 converter is still on its way, but I expected to at least see some scrambled data.

Thanks for any assistance

jbuehl, 2020-03-29:
What may be happening is that your inverter is trying to connect to a port other than the ones that semonitor.py is listening on by default (22222, 22221, 80).  You can find out what that is by capturing the data on the network and looking at it with Wireshark.  If it is expecting to use SSL then semonitor.py definitely won't work.

Note that in the latest version of the code, the `-n` option is not working.  See issue #148

hydrael, 2020-03-29:
I was listening on 22222 and 22221...I had a conflict on port 80, which is why I couldn't use that.
`prodssl.solaredge.com` at least has port 22222 open.

But I'll verify what's happening by looking at a wireshark dump.

If it turns out my inverter wants to use SSL, would usage of active RS485 be an option?

jbuehl, 2020-03-29:
I don't know anything about the capabilities of the newer inverters and firmware versions.  Look through the issues here and maybe you can find the answer.

hydrael, 2020-03-29:
Alright, thanks

Millox, 2020-04-20:
I can now confirm this. The newest firmware, from april 2020, uses SSL by default and only contacts prodssl.solaredge.com. I have no trace of communication to any of the prod-prod3-servers at the previously mentioned ports.

cristianlivella, 2020-04-20:
> I can now confirm this. The newest firmware, from april 2020, uses SSL by default and only contacts prodssl.solaredge.com. I have no trace of communication to any of the prod-prod3-servers at the previously mentioned ports.

Does the web interface of the inverter still work? There seems to be a problem (or an intentional block) with the last updates: https://github.com/drobtravels/solaredge-local/issues/24#issuecomment-616518724

Millox, 2020-04-20:
Yes, the local web interface still works, connecting to the built in AP. Remotely (i.e. using the lan interface) does not work. nmap:ing the device yields ports 80 and 8080 filtered and the rest closed.

borg16, 2021-02-25:
Sorry for bringing this 10 month-old issue back: I have a new SolarEdge _Three Phase System with SetApp Configuration_ and have the SolarEdge monitoring running. It connects only to prodssl.solaredge.com and only via port 443. Can you confirm that this will never allow semonitor ethernet passive and probably never ethernet active? So no use for me to retrieve the encryption key, right?