jbuehl/solaredge #150: Optimizer data via modbus

marcpuca, 2020-04-06:
Sorry that I used the issue submission to ask this question but couldnt find any contact for this.

Can the optimizer data be fetched via modbus and if so could you kindly provide details on the registers that hold the data (I have written a java client that fetches register data using modbus tcp) 

Thanks
Marc

kingfisher63, 2020-04-06:
Currently not (firmware 4.8.28). When I scan the Modbus interface on my SE4000H, I see only 2 SunSpec models:

- 1: Common
- 101: Inverter (Single Phase)

crazyelectron-io, 2020-04-06:
> I see only 2 SunSpec models:
>
> 1: Common
> 101: Inverter (Single Phase)

Similar for my setup (SE7K): 1: Common, 103: Inverter (3-phase).

Sunspec does not support optimizer data because that is SolarEdge proprietary.

So far, I have been able to get the day, month, year, and total energy (Wh) using the RS232 interface (in real-time) but have not yet found a way to get the optimizer data. Nor the real-time power production (W) via RS232, only via RS485.

Using the RS485 interface configured for the SE protocol, as slave, should theoretically provide all the data, but I have not yet succeeded in getting that out of the RS485 interface.

crazyelectron-io, 2020-04-07:
> model 502 (solar module model supporting DC-DC converter) should do

I should have had a better look at the Sunspec specifications... 😎

jbuehl, 2020-04-07:
A SolarEdge optimizer is a DC-DC converter, but there is more to it.  The inverter communicates with the optimizers individually over the DC power lines so that the voltage of the string is always close to 340v, which is the most efficient value for a 240v inverter.  That's how if a panel gets partially shaded, all the optimizers in the string can adjust their output voltages to compensate.  This implementation and the  communication protocol is proprietary to SolarEdge, so unless they license it to another manufacturer you won't likely see the ability to report individual optimizer (panel) data in SunSpec.  That is why this project came about in the first place.

kingfisher63, 2020-04-16:
The idea is that the inverter exposes optimizer data using type 502 SunSpec model blocks (one block per optimizer) in exactly the same way as SolarEdge is already doing for SolarEdge energy meters attached to the inverter (see [this document](https://www.solaredge.com/sites/default/files/sunspec-implementation-technical-note.pdf)). The inverter would still be the only device communicating with the optimizers, so there will be no communication or licensing issues.

Of course this will only become reality if SolarEdge product management thinks that this is important enough to allocate some engineering resources. IMHO this is unlikely to happen short term.

geertn444, 2020-11-08:
@kingfisher: +1 count me in. I am reading the general data, but would like to see also optimizer date (not real time but latest status) also W per panel for example. I am also looking to see if it is possible to measure string current per string. I have two strings, but modbus only returns a single "DC current" value. So (?) is this an average then ? don't know...