jbuehl/solaredge #156: Could firmware update of inverter SE4K conflict with old semonitor.py via RS485?

ethpi, 2020-09-26:
Hi,

since 3 years I run semonitor.py over RS485 in active mode with my Solaredge SE4K 3 phase inverter and everything works perfectly. So I disconnected the inverter from the internet and never updated semonitor.py. Everything is allright.

But actually one of my moduls/optimizers produces only 10% of its normal power. I contacted my local craftsman who installed the system. He wants to have look at the monitoring panel of solaredge.com, but there is no actual data because I ran the data collection via RS485 to my own MySQL database for years. 
I don't have a problem to reconnect my inverter to the internet and loose a few days of data logging in my database. But I'm afraid that solaredge updates the firmware of my inverter and a eventual change of the data protocol or anything else causes a malfunction of the data logging with my old version of semonitor.py. I'd like to avoid this situation. Never cange a running system.

Does anybody know if there is a SE4K firmware update since 2015 that causes a failure of the data logging via RS485 or Ethernet?

Thanks for answers.

jbuehl, 2020-09-26:
I am reluctant to connect my inverters to the internet for the same reason.  I haven't followed the discussions closely, but I get the impression that there have been firmware updates that will cause problems with software from this project.  There shouldn't be anything on the SolarEdge monitoring panel that you don't have in the data you have collected.  If there is a way that you can give your installer access to your data and convince him that is is just as valid as what SolarEdge would have had then he should be able to get what he needs.

ethpi, 2020-09-27:
Ok - I won't reconnect the inverter to the internet. It's no problem to share my data with my installer, because I have a nice script that produces html to show the data in a browser.

closed 2021-11-19 by ethpi