jbuehl/solaredge #157: No data on RS485 passive

dennisfisch, 2020-10-27:
Hey there,
my setup is
SE-WND-3Y400-MB (Energy Meter)
|
SE3000H (Inverter with Display, ZigBee, Ethernet, 2x RS485 (2nd is unconnected), no SetApp, no RS232 that i can find on the PCB)
|
StoreEdge
|
LG Resu7H

I have connected my Raspberry PI Zero W with the Energy Meter. I disabled the termination resistor on it and installed a 130R along with the wires into the WaveShare RS485 / CAN HAT.

This same HAT works perfectly fine talking to my HeatPump, but i cannot for the life of me get any useful data using semonitor.py as it just continues to hang indefinitely:
`pi@raspberrypi:~/solaredge $ python semonitor.py -vvvv -t 4 -d stdout /dev/ttyAMA0 -b 9600
append: False
baudrate: 9600
commands:
datasource: /dev/ttyAMA0
follow: False
following: True
interface: None
keyfile: None
logfile: stdout
master: False
outfile: stdout
ports: 22222,22221,80
record: None
slaves:
type: 4
updatefile: None
verbose: 4
xerror: False
opening /dev/ttyAMA0`
I've monitored the serial port using SCREEN (screen /dev/ttyAMA0 9600) and sure enough, there is data coming into that port.
The input matches the flashing LED of the energy meter and stops as soon as i yank out a wire so its definitely coming through the serial port.
Data is visible and consistent only at baudrate 9600, getting less and less visible until 57600, after which nothing comes though at the default of 115200, leading me to believe it has been configured to be 9600 for my installation.

I then tried to record the incoming data with semonitor using the -r flag which produces a file with binary content.
I cannot really interpret the modbus frames in there, or even what to look for. Is there anything that I'm missing?

At one point, the program aborted on me with this message:
`pi@raspberrypi:~/solaredge $ python semonitor.py -vvvv -t 4 -d stdout /dev/ttyAMA0 -r out.txt -b 9600
append: False
baudrate: 9600
commands:
datasource: /dev/ttyAMA0
follow: False
following: True
interface: None
keyfile: None
logfile: stdout
master: False
outfile: stdout
ports: 22222,22221,80
record: out.txt
slaves:
type: 4
updatefile: None
verbose: 4
xerror: False
opening /dev/ttyAMA0
Exception while reading data: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
/dev/ttyAMA0 --> message: 1 length: 1006
data:       12 34 56 79 86 7e 60 fe 18 80 1e 80 60 78 80 66
data:       e6 80 80 18 80 1e 80 e0 80 80 f8 86 1e 98 7e 86
data:       f8 98 f8 e6 18 98 18 86 86 78 fe 1e 98 e0 60 7e
data:       e6 60 80 1e e0 1e 9e e6 80 e0 e6 e6 80 e0 e6 66
data:       e6 f8 06 80 06 80 e6 80 e0 e6 06 18 60 f8 18 80
data:       1e 80 60 78 80 66 e6 80 80 18 80 1e 80 e0 80 9e
data:       86 1e 9e 1e 98 fe 60 80 78 86 86 80 06 18 1e 98
data:       f8 f8 1e 98 e0 60 f8 e6 60 80 1e e0 1e 9e e6 80
data:       e0 e6 e6 80 e0 e6 66 e6 f8 06 80 06 80 e6 80 e0
data:       e6 fe f8 e0 f8 18 80 1e 80 60 78 80 66 e6 80 80
data:       18 80 1e 80 e0 80 e0 60 60 e6 80 86 1e e6 e0 86
data:       18 18 98 1e 86 86 98 86 1e 98 e0 60 78 78 7e fe
data:       e6 60 80 1e e0 1e 9e e6 80 e0 e6 e6 80 e0 e6 66
data:       e6 f8 06 80 06 80 e6 80 e0 e6 06 18 9e 18 80 1e
data:       80 60 78 80 66 e6 80 80 18 80 1e 80 e0 80 86 fe
data:       78 98 1e 98 e0 60 86 86 66 06 66 f8 60 06 86 60
data:       e6 80 e6 1e 66 e6 86 9e 86 fe e0 80 60 f8 60 78
data:       1e 06 7e 60 fe 18 80 1e 80 60 78 80 66 e6 80 80
data:       18 80 1e 80 e0 80 98 1e 66 e6 80 86 1e 06 9e 80
data:       60 06 66 7e 98 e0 9e fe 1e 98 e6 e6 60 80 1e e0
data:       1e 9e e6 80 e0 e6 e6 80 e0 66 e6 f8 06 80 06 80
data:       e6 80 e0 e6 06 18 60 f8 18 80 1e 80 60 78 80 66
data:       e6 80 80 18 80 1e 80 e0 80 fe 98 e0 f8 1e 98 e0
data:       60 e6 66 86 86 80 86 18 9e e6 06 e6 80 66 e6 60
data:       80 1e e0 1e 9e e6 80 e0 e6 e6 80 66 e6 f8 06 80
data:       06 98 06 18 80 1e 80 78 80 66 e6 80 80 18 80 1e
data:       80 e0 80 86 fe fe f8 1e 98 98 60 f8 98 7e 06 66
data:       60 e6 e0 fe 98 80 06 60 7e 60 7e e6 60 80 1e e0
data:       1e 9e e6 80 e0 e6 e6 80 66 e6 f8 06 80 06 80 e6
data:       80 e0 e6 98 78 66 18 80 1e 80 78 80 66 e6 80 80
data:       18 80 1e 80 e0 80 86 78 80 06 1e 86 06 18 98 86
data:       fe e0 80 60 f8 60 78 1e 78 fe 7e 60 fe 18 80 1e
data:       80 60 78 80 66 e6 80 80 18 80 1e 80 e0 80 e6 60
data:       f8 1e 98 86 60 18 9e 66 18 98 98 60 1e 06 e6 1e
data:       98 98 60 1e e6 f8 fe 18 9e 66 78 80 86 66 18 fe
data:       18 80 1e 80 60 78 80 66 e6 80 80 18 80 1e 80 e0
data:       80 78 60 18 98 80 66 1e 66 fe 9e 78 06 66 06 18
data:       60 98 7e 98 80 66 60 66 06 18 e6 60 80 1e e0 1e
data:       9e e6 80 e0 e6 e6 80 e0 66 e6 f8 06 80 06 80 e6
data:       80 e0 e6 9e 66 fe 18 80 1e 80 60 78 80 66 e6 80
data:       80 18 80 1e 80 e0 80 78 66 1e 1e 98 18 86 fe 80
data:       18 98 80 86 f8 80 98 86 1e 98 06 e6 60 80 1e e0
data:       1e 9e e6 80 e0 e6 e6 80 e0 66 e6 f8 06 80 06 80
data:       e6 80 e0 e6 98 78 66 18 80 1e 80 60 78 80 66 e6
data:       80 80 18 80 1e 80 e0 80 9e e0 80 60 80 86 1e 06
data:       9e 98 1e 80 86 60 06 7e 18 e0 1e 98 fe 60 e6 60
data:       80 1e e0 1e 9e e6 80 e0 e6 e6 80 e0 66 e6 f8 06
data:       80 06 80 e6 80 e0 e6 66 9e e0 e0 18 80 1e 80 78
data:       80 66 e6 80 80 18 80 1e 80 e0 80 fe 60 f8 1e 98
data:       86 60 7e 18 60 18 98 f8 60 66 66 60 9e 80 86 86
data:       fe e0 80 f8 60 78 1e 06 7e 60 fe 18 80 1e 80 78
data:       80 66 e6 80 80 18 80 1e 80 e0 80 98 78 78 78 9e
data:       9e e6 60 80 1e e0 1e 9e e6 80 e0 e6 e6 80 66 e6
data:       f8 06 06 18 fe 18 80 1e 80 78 80 66 e6 80 80 18
data:       80 1e 80 e0 80 60 fe 06 86 1e 98 60 86 18 fe 86
data:       f8 18 98 18 86 60 98 e0 1e 98 e0 60 fe 06 f8 e6
data:       60 80 1e e0 1e 9e e6 80 e0 e6 e6 80 e0 60 e6 f8
data:       06 80 06 80 e6 80 e0 e6 fe 80 e6 fe 18 80 1e 80
data:       60 78 80 66 e6 80 80 18 80 1e 80 e0 80 86 1e e6
data:       80 86 06 18 98 fe 98 18 98 e6 60 9e 1e e6 80 e6
data:       1e 66 66 e6 60 80 1e e0 1e 9e e6 80 e0 e6 e6 80
data:       e0 66 e6 f8 06 80 06 80 e6 80 e0 e6 06 06 e0 18
data:       80 1e 80 78 7e 60 18 66 f8 f8 18 80 1e 80
closing /dev/ttyAMA0
closing out.txt
closing <stdout>`

cwsmith, 2020-10-27:
Do you know what firmware your inverter is running?

According to this doc
https://www.solaredge.com/sites/default/files/sunspec-implementation-technical-note.pdf
, under SunSpec Supported Inverters, 
 you'll need > 3.xxxx for SunSpec support.

dennisfisch, 2020-10-27:
To follow up on this, i obtained an RS485 USB Converter and plugged it into my laptop.
Together with SerialPortMonitor (unfortunately not free....) i got it to list some interactions:
[image]

Seems the PI is not able to keep up using the WaveShare RS485 HAT. Unfortunately it is not reporting any data using the USB dongle either.

dennisfisch, 2020-10-27:
> Do you know what firmware your inverter is running?
> 
> According to this doc
> https://www.solaredge.com/sites/default/files/sunspec-implementation-technical-note.pdf
> , under SunSpec Supported Inverters,
> you'll need > 3.xxxx for SunSpec support.

Thanks for your response. I just checked it on the display as described in the Spec you mentioned and the version is reported as 3.2525

jbuehl, 2020-10-27:
I'm not familiar with that hardware, but if it gets data from the inverter in the SunSpec format it isn't going to work with semonitor.py.  This software is for reading data in the proprietary SolarEdge format.

dennisfisch, 2020-10-27:
Soooo.. by the looks of it this is the case here? In that case i'm sorry for opening an issue for what is obviosly my own fault.
I do not have adminstrative access to the inverter and cannot change any of its settings. 

I guess i'll have to look into a different solution.

cwsmith, 2020-10-28:
A few sunspec parsers that look interesting:
https://github.com/nmakel/solaredge_modbus
https://github.com/tjko/sunspec-monitor

nflch, 2022-02-08:
> I'm not familiar with that hardware, but if it gets data from the inverter in the SunSpec format it isn't going to work with semonitor.py. This software is for reading data in the proprietary SolarEdge format.

My Inverter does not have RS232 so I tried the RS485 interface. After one day work I get this tracker.

@jbuehl 
PLEASE add in the description (README.md in RS485 part), that the protocol _must_ be the proprietary SolarEdge format and _not_ the modbus format.
Thank you a lot