jbuehl/solaredge #167: Multiple Connections

gamblor999, 2022-01-28:
@jbuehl  hopefully a basic question!
If I connect an RS485 to the inverter set as slave and use semonitor.py to collect data via a rpi, can I still have my Ethernet connection reporting to the online Solaredge monitoring platform simultaneously? Or is it one or the other ?

InFerYes, 2022-07-24:
You could always just connect it to your network and report to SolarEdge and at the same time listen to the packets that are sent along the network with something like Wireshark and filter the packages coming from the device.

> Traffic may be captured to a pcap file using wireshark or tcpdump or some other program.
The resulting file must be processed through seextract.py before it is fed into semonitor.py. In this mode, semonitor.py is reading data from a file or stdin and isn't actually reading from the network directly.

eschwim, 2022-10-06:
AFAICT it is either one or the other (at least on my setapp-enabled inverter I can only choose one "monitoring connection").  When I had my rpi+rs485 semonitor configured as slave, I wasn't getting anything across the wire other than basic RS-485 keepalive traffic.  Ditto when I had the rpi configured as master, but had the inverter reporting monitoring data over ethernet.

And I believe that, as of now, monitoring via ethernet proxying/capture is a non-starter, as inverters with a recent firmware version are now using TLS-backed encryption (instead of the previously implemented hey-key-based encryption, which was trivially decyptable) to transmit inverter+optimizer data to the Solaredge portal.  Combined with a PKI/root certs on the inverter that you can't tamper with, and server-certificate validation, this means that we are effectively locked out of capturing data via the network.

I discovered this when my old SE11400H-US inverter (one of the ones with an LCD screen, which did not support setapp) was replaced with one of the new setapp-only models due to an RMA.  My previously working ethernet-based implementation no longer worked, and after a few tcpdumps it became pretty clear that there is no longer any non-TLS traffic being sent between the inverter and the Solaredge portal.

TL;DR RS485 is the only way to capture optimizer and inverter production data now.