jbuehl/solaredge #175: Can't get useful data from SE10000H via RS-485, Unknown Function

cschuhen, 2022-12-22:
Hello,

I have an SE10000H, connecting via RS-485, running latest firmware.

Inverter is set to SolarEdge Follower with ID of 1.

Python version 3.9.2 running on a Raspberry PI 2+.

I seem to getting some packets back but none that are recognized. I am getting these functions: 0302, 039a, 4288, 428a.

I also see this message on occasion: RS485 master ack timeout

I have tried many variations of the command for example:

sudo ./semonitor.py -t 4 -o 20221223.json -m -s 74047697  -d stdout -vvvv /dev/ttyUSB0 -b 115200 2>&1 | tee dump4.data

cschuhen, 2022-12-22:
Here is some example data:

append: False
baudrate: 115200
commands: 
datasource: /dev/ttyUSB0
follow: False
following: True
interface: None
keyfile: None
logfile: stdout
master: True
outfile: 20221223.json
ports: 22222,22221,80
record: None
slaves: 74047697
type: 4
updatefile: None
verbose: 4
xerror: False
opening /dev/ttyUSB0
starting read thread
starting master thread
dataLen:    0000
dataLenInv: ffff
sequence:   2007
source:     fffffffe
dest:       74047697
function:   0302
/dev/ttyUSB0 <-- message: 1 length: 22
data:       12 34 56 79 00 00 ff ff 07 20 fe ff ff ff 97 76
data:       04 74 02 03 92 e7
b''
/dev/ttyUSB0 --> message: 2 length: 251
data:       12 34 56 79 e5 00 1a ff 30 bc 97 76 04 74 fe ff
data:       ff ff 88 42 02 00 01 00 15 00 49 6e 76 50 61 63
data:       6b 2d 37 34 38 34 37 36 39 37 00 03 00 ca 00 0a
data:       10 49 6e 76 50 61 63 6b 2d 37 34 38 34 37 36 39
data:       37 10 00 1a 33 0a 02 08 00 12 02 08 00 1a 05 0d
data:       aa 4f 19 45 22 05 0d ca 0b b5 44 2a 05 08 f1 ae
data:       88 13 32 02 08 00 3a 02 08 00 42 05 0d 00 00 00
data:       00 4a 05 0d 00 00 00 00 22 12 0a 00 12 00 1a 00
data:       22 00 2a 08 31 2e 30 2e 31 36 31 34 2a 6b 0a 02
data:       08 01 12 05 0d 00 40 1c 46 1a 05 0d 00 40 1c 46
data:       22 05 0d 00 00 00 00 2a 02 08 00 32 05 0d 00 00
data:       00 00 3a 02 08 00 42 02 08 00 4a 02 08 00 52 02
data:       08 00 5a 02 08 00 62 02 08 00 6a 10 0a 02 08 00
data:       12 02 08 00 1a 02 08 00 22 02 08 00 70 00 7a 00
data:       82 01 00 8a 01 13 0a 02 08 00 12 05 0d 00 00 00
data:       00 1a 02 08 00 22 02 08 00 14 3a
Could not open last0503.msg file, not loading
No data read from last0503.msg. Not loading
dataLen:    00e5
dataLenInv: ff1a
sequence:   bc30
source:     74047697
dest:       fffffffe
function:   4288
Unknown function 0x4288
/dev/ttyUSB0 --> message: 3 length: 251
data:       12 34 56 79 e5 00 1a ff a4 bc 97 76 04 74 fe ff
data:       ff ff 88 42 02 00 01 00 15 00 49 6e 76 50 61 63
data:       6b 2d 37 34 38 34 37 36 39 37 00 03 00 ca 00 0a
data:       10 49 6e 76 50 61 63 6b 2d 37 34 38 34 37 36 39
data:       37 10 00 1a 33 0a 02 08 00 12 02 08 00 1a 05 0d
data:       aa 4f 19 45 22 05 0d ca 0b b5 44 2a 05 08 f1 ae
data:       88 13 32 02 08 00 3a 02 08 00 42 05 0d 00 00 00
data:       00 4a 05 0d 00 00 00 00 22 12 0a 00 12 00 1a 00
data:       22 00 2a 08 31 2e 30 2e 31 36 31 34 2a 6b 0a 02
data:       08 01 12 05 0d 00 40 1c 46 1a 05 0d 00 40 1c 46
data:       22 05 0d 00 00 00 00 2a 02 08 00 32 05 0d 00 00
data:       00 00 3a 02 08 00 42 02 08 00 4a 02 08 00 52 02
data:       08 00 5a 02 08 00 62 02 08 00 6a 10 0a 02 08 00
data:       12 02 08 00 1a 02 08 00 22 02 08 00 70 00 7a 00
data:       82 01 00 8a 01 13 0a 02 08 00 12 05 0d 00 00 00
data:       00 1a 02 08 00 22 02 08 00 ef 55
dataLen:    00e5
dataLenInv: ff1a
sequence:   bca4
source:     74047697
dest:       fffffffe
function:   4288
Unknown function 0x4288
/dev/ttyUSB0 --> message: 4 length: 251
data:       12 34 56 79 e5 00 1a ff 22 bd 97 76 04 74 fe ff
data:       ff ff 88 42 02 00 01 00 15 00 49 6e 76 50 61 63
data:       6b 2d 37 34 38 34 37 36 39 37 00 03 00 ca 00 0a
data:       10 49 6e 76 50 61 63 6b 2d 37 34 38 34 37 36 39
data:       37 10 00 1a 33 0a 02 08 00 12 02 08 00 1a 05 0d
data:       aa 4f 19 45 22 05 0d ca 0b b5 44 2a 05 08 f1 ae
data:       88 13 32 02 08 00 3a 02 08 00 42 05 0d 00 00 00
data:       00 4a 05 0d 00 00 00 00 22 12 0a 00 12 00 1a 00
data:       22 00 2a 08 31 2e 30 2e 31 36 31 34 2a 6b 0a 02
data:       08 01 12 05 0d 00 40 1c 46 1a 05 0d 00 40 1c 46
data:       22 05 0d 00 00 00 00 2a 02 08 00 32 05 0d 00 00
data:       00 00 3a 02 08 00 42 02 08 00 4a 02 08 00 52 02
data:       08 00 5a 02 08 00 62 02 08 00 6a 10 0a 02 08 00
data:       12 02 08 00 1a 02 08 00 22 02 08 00 70 00 7a 00
data:       82 01 00 8a 01 13 0a 02 08 00 12 05 0d 00 00 00
data:       00 1a 02 08 00 22 02 08 00 1b 07
dataLen:    00e5
dataLenInv: ff1a
sequence:   bd22
source:     74047697
dest:       fffffffe
function:   4288
Unknown function 0x4288
RS485 master ack timeout
dataLen:    0000
dataLenInv: ffff
sequence:   2008
source:     fffffffe
dest:       74047697
function:   0302
/dev/ttyUSB0 <-- message: 2 length: 22
data:       12 34 56 79 00 00 ff ff 08 20 fe ff ff ff 97 76
data:       04 74 02 03 a2 f3

[dump4.txt](https://github.com/jbuehl/solaredge/files/10290337/dump4.txt)

cschuhen, 2022-12-22:
Ok, so I went to go check my firmware version to add do my post here. And noticed another configuration option.

In Monitoring Communication, I needed to also select RS485-1 as the connection type.

closed 2022-12-22 by cschuhen

scr34m, 2023-09-17:
What was the solution in the configuration you had to set?
I have the same problem, receiving 4288 and 428a functions

scr34m, 2023-09-17:
How stable is the connection for you, i've changed the monitoring to RS485 and i readed few data and the led turned blue but after few min the blue led turn off and receiving only 4288, 428a data.