jbuehl/solaredge #177: Just a little help RS485 please

eriksl, 2023-05-26:
Hi there,

I am now more or less successfully monitoring from the RS485 interface (inverter is slave, semonitor is master). There are a few things that really puzzle me.

1)  I am only getting old data, data from days ago... Does the inverter store some amount of unfetched data?
2) the process is going very slowly, it takes about a minute to get the next message, in the meantime messages are sent but apparently no response from the inverter. I also get "RS485 master ack timeout" a lot.

It looks like everything is setup ok, otherwise I wouldn't receive any data at all I guess. First I was trying to get connect the RS485 cable to my server, which involves about 12 meters of UTP cable. So I thought that might be the problem. So now I connect a laptop next to the inverter, cable is only 1 m long and now I am having exactly the same issue.

I also swapped the RS485 interfaces (computer side), I have one with a USB interface and one with an RS232 interface. This appears to have no effect at all. One of them has the option to set a terminating resistance of 120 Ohms, which I did and that also does not make a difference.

The UTP cables I wired like: A+ on orange, B- on orange-white, all other wires to ground.

So I really don't have  clue here, can someone help me here?

eriksl, 2023-05-26:
Ah thanks very much! That gives me hope it will going to work some time. So it's normal it can take 30 seconds between any messages?

At the moment I am not getting any data though, just "filler" messages. Wonder what that's supposed to mean! Just restarted semonitor and now I am again getting messages with real data. Interesting.

If you say "it's not realtime", will it be possible to have data from e.g. the last 5 minutes or not even that?

Do you have experience using the modbus interface (last resort if this isn't going to work, I know I won't have optimizer data then...)?

jbuehl, 2023-05-26:
A message every 30 seconds or so from each inverter is about what I usually see.  I'm not sure how exactly they decide when to send a 0x0500 message which contains inverter data and some, but not all, optimizer data.  Inverter data should be from within the last minute, but data for a given optimizer might be updated less often.  Expecting to see everything updated every 5 minutes during daylight is reasonable.  At night there are just periodic keepalive messages.

I haven't tried to do anything with the modbus interface.

eriksl, 2023-05-26:
Ok, sounds good, thanks!

closed 2023-09-09 by eriksl