jbuehl/solaredge #178: Remote serial converter

eriksl, 2023-05-26:
Hi!

Not an issue but a feature request, I am not sure if this is easy to implement. I am a noob at Python otherwise I'd probably do it myself! The request is to be able to be able to use a network socket instead of a serial port for rs485 (would be nice if both tcp and udp would be supported). My esp8266's have serial port forwarding, then it's only a matter of connecting a rs232<->rs485 converter and I can do away with the energy hogging laptop for this purpose.

Would be very nice, thanks in advance!

jbuehl, 2023-05-26:
Depending on what you want to run in your esp8266 there are a couple of ways to do it.  You could use a utility such as [socat](https://stackoverflow.com/questions/22624653/create-a-virtual-serial-port-connection-over-tcp) that creates a virtual serial port on the device that you are running semonitor and forwards the messages to a port on the esp.  Or you could run semonitor in network mode (-t n) and write code for the esp which would make a tcp connection to port 22222 on the semonitor device and forward the messages that way.

eriksl, 2023-05-26:
That's a great idea. I am familiar with SOCAT but I didn't realize it can emulate a serial port.

Are the messages on port 22222 exactly the same?

BTW the firmware on the ESP8266 is something I develop myself, so there are some possibilities to be creative there if necessary :-) The only worry I have is that I currently do not allocate very much ram for serial port buffering, so it may not be able to keep up with the rather large messages (it seems). We'll cross that bridge when we come to it, the UART won't be the problem, it can handle up to 80 Mbps...

If I get the UTP cable (12 meters) working after all, this all may not be necessary. Fingers crossed. I just read something about termination on two sides, I only did one side...

jbuehl, 2023-05-27:
After thinking about this more, the method using the virtual serial port is definitely the easier way to go as it would not require any code changes, whereas the other way would require either semonitor to add the RS485 master grant messages to network mode or for the ESP to handle those messages, which would be non-trivial.

I would suggest using UDP for the network connection rather than TCP because the solaredge protocol already handles error detection and retries, so the extra reliability of TCP is redundant.  Also I don't know what would happen if socat loses a socket connection.  You might wind up having to wrap it in something to handle reconnections.

eriksl, 2023-05-28:
In the end I might be implementing the whole protocol in the ESP itself, that would be most robust I guess. At the moment I can use both UDP and TCP and I found TCP isn't really that heavier than UDP on the ESP. It has the advantage though that record boundaries are preserved and inside lwip the data isn't queue but sent immediately.

For the moment I am now using a small laptop for this purpose (which runs semonitor itself). I am not yet sure which way to go. My findings until now:

- snooping on messages to the solaredge servers (1) is no longer possible because they're using tls
- snooping on messages to the solaredge servers (2) is no longer possible because they're using tls and sslsplit doesn't work because they check the validity of the CA
- retrieving records using RS 485 is a drag. It looks like the retrieval rate is slower than new records are created, so I'll never catch up.

Also I keep getting master ack timeouts. I already tried several workarounds, none work:
- adding/removing terminating resistors
- changing RS 485 interface on the laptop side
- changing RS 485 interface on inverter side

The only things I can think of
- ground loop, try with the ground connections disconnected
- voltage on A/B from computer to inverter is too low to be reliably detected by the inverter, I am not sure what the threshold is there, I am not sure how to fix that
- maybe increasing the master ack timeout?

In the meantime I am fetching the "general" data using tcp modbus and that really works like a charm. Request info and get it back instantly, no delay whatsoever. If only they had added optimiser data there...

Thanks for sharing your thoughts!

jbuehl, 2023-05-28:
I'm inclined to think it's a software issue rather than a RS-485 electrical problem.  I would expect that you would be seeing a lot of messages about checksum and length errors if it were the latter.  The master ack timeout means that semonitor just isn't seeing any response when it polls the inverter.  It's possible that the hardware you are running semonitor on is too slow to catch the response from the inverter.  I wouldn't rule out a bug in semonitor either.  You could try decreasing the value of MASTER_MSG_INTERVAL in semonitor.py to cause it to poll more often which should speed things up but not necessarily fix the timeout problem.

Can you run semonitor with the -vvv option and post the debug output here?

eriksl, 2023-05-29:
Thanks for your interest!

I don't think "slow" hardware is the issue, I tried on both a nettop (atom CPU) and a laptop (core i5 CPU).

I am now running with -vvvv (four times, I will append some of the output (it has been running for over a day now...)

Interesting option this MASTER_MSG_INTERVAL, will try!

At the moment I also have another issue (which is not really semonitor related: my inverter keeps resetting the time/date to 2019-12-12/00:00. I now have one day worth of usable data, which already show very interesting things, I might not be persueing this path in the end, if I can't get it working properly. The modbus interface works really well and gives all information except for the individual optimisers.

```
dataLen:    0000
dataLenInv: ffff
sequence:   0549
source:     fffffffe
dest:       7415f893
function:   0080
/dev/ttyUSB0 <-- message: 3767 length: 22
data:       12 34 56 79 00 00 ff ff 49 05 fe ff ff ff 93 f8
data:       15 74 80 00 68 06
RS485 master ack timeout
dataLen:    0000
dataLenInv: ffff
sequence:   2f47
source:     fffffffe
dest:       00000001
function:   0302
/dev/ttyUSB0 <-- message: 3768 length: 22
data:       12 34 56 79 00 00 ff ff 47 2f fe ff ff ff 01 00
data:       00 00 02 03 d2 22
RS485 master ack timeout
dataLen:    0000
dataLenInv: ffff
sequence:   2f48
source:     fffffffe
dest:       00000001
function:   0302
/dev/ttyUSB0 <-- message: 3769 length: 22
data:       12 34 56 79 00 00 ff ff 48 2f fe ff ff ff 01 00
data:       00 00 02 03 e2 36
RS485 master ack timeout
dataLen:    0000
dataLenInv: ffff
sequence:   2f49
source:     fffffffe
dest:       00000001
function:   0302
/dev/ttyUSB0 <-- message: 3770 length: 22
data:       12 34 56 79 00 00 ff ff 49 2f fe ff ff ff 01 00
data:       00 00 02 03 e6 ca
/dev/ttyUSB0 --> message: 1963 length: 81
data:       12 34 56 79 3b 00 c4 ff ff ff 93 f8 15 74 fe ff
data:       ff ff c2 03 00 00 33 00 a7 8d a1 5d 00 00 00 00
data:       3d c0 b4 44 3d c0 b4 44 ff ff 7f ff ff ff 7f ff
data:       ff ff 7f ff ff ff 7f ff ff ff 7f ff ff ff 7f ff
data:       ff ff 7f ff ff ff 7f ff 00 00 00 00 00 00 00 88
data:       92
dataLen:    003b
dataLenInv: ffc4
sequence:   ffff
source:     7415f893
dest:       fffffffe
function:   03c2
Unknown function 0x03c2
/dev/ttyUSB0 --> message: 1964 length: 22
data:       12 34 56 79 00 00 ff ff 4a 05 93 f8 15 74 fe ff
data:       ff ff 00 05 ce 3e
dataLen:    0000
dataLenInv: ffff
sequence:   054a
[... 687 lines cut ...]
    Pmax : 11000.0
    Pac : 2429.16064453125
meters_0x0022:  7415F893 type: 0022 len: 003a
    meters_0x0022 : {'7415F893': {'9_PVProduction': {'seType': '0x0022', 'seId': '7415F893', 'devLen': 58, 'devType': 'meters_0x0022', 'dateTime': 1570868947, 'Date': '2019-10-12', 'Time': '10:29:07', 'recType': 9, 'onlyIntervalData': 1, 'TotalE2Grid': 0, 'AlwaysZero_off10_int2': 0, 'Flag_off12_hex': '00 80', 'TotalEfromGrid': 0, 'AlwaysZero_off18_int2': 0, 'Flag_off20_hex': '00 80', 'Totaloff22_int4': 0, 'AlwaysZero_off26_int2': 0, 'Flag_off28_hex': '00 80', 'Totaloff30_int4': 0, 'AlwaysZero_off34_int2': 0, 'Flag_off36_hex': '00 80', 'Interval': 300, 'E2X': 171, 'EfromX': 0, 'P2X': 2429.16064453125, 'PfromX': nan}}}
{"events": {}, "inverters": {"7415F893": {"Date": "2019-10-12", "Eac": 171.0, "Eday": 3692.275634765625, "Etot": 207073.0, "Freq": 50.005226135253906, "ID": "7415F893", "Iac": 10.313653945922852, "Interval": 300, "Pac": 2429.16064453125, "Pmax": 11000.0, "Temp": 41.996212005615234, "Time": "10:29:07", "Uptime": 35491, "Vac": 236.00656127929688, "Vdc": 375.98248291015625}}, "meters_0x0022": {"7415F893": {"9_PVProduction": {"AlwaysZero_off10_int2": 0, "AlwaysZero_off18_int2": 0, "AlwaysZero_off26_int2": 0, "AlwaysZero_off34_int2": 0, "Date": "2019-10-12", "E2X": 171, "EfromX": 0, "Flag_off12_hex": "00 80", "Flag_off20_hex": "00 80", "Flag_off28_hex": "00 80", "Flag_off36_hex": "00 80", "Interval": 300, "P2X": 2429.16064453125, "PfromX": NaN, "Time": "10:29:07", "TotalE2Grid": 0, "TotalEfromGrid": 0, "Totaloff22_int4": 0, "Totaloff30_int4": 0, "dateTime": 1570868947, "devLen": 58, "devType": "meters_0x0022", "onlyIntervalData": 1, "recType": 9, "seId": "7415F893", "seType": "0x0022"}}}, "optimizers": {"153F9EBE": {"Date": "2019-10-12", "Eday": 0, "ID": "153F9EBE", "Imod": 7.8187500000000005, "Inverter": 0, "Temp": 0, "Time": "10:26:45", "Uptime": 14313, "Vmod": 30.75, "Vopt": 49.375}, "15674F79": {"Date": "2019-10-12", "Eday": 0, "ID": "15674F79", "Imod": 2.4125, "Inverter": 0, "Temp": 0, "Time": "10:29:03", "Uptime": 14452, "Vmod": 30.75, "Vopt": 11.125}, "15675DD8": {"Date": "2019-10-12", "Eday": 0, "ID": "15675DD8", "Imod": 2.11875, "Inverter": 0, "Temp": 0, "Time": "10:27:34", "Uptime": 14363, "Vmod": 31.0, "Vopt": 10.0}, "1569472A": {"Date": "2019-10-12", "Eday": 0, "ID": "1569472A", "Imod": 2.0625, "Inverter": 0, "Temp": 0, "Time": "10:27:22", "Uptime": 14351, "Vmod": 31.25, "Vopt": 10.625}, "156C67FB": {"Date": "2019-10-12", "Eday": 0, "ID": "156C67FB", "Imod": 2.24375, "Inverter": 0, "Temp": 0, "Time": "10:28:25", "Uptime": 14414, "Vmod": 31.25, "Vopt": 11.0}, "173675DF": {"Date": "2019-10-12", "Eday": 0, "ID": "173675DF", "Imod": 11.30625, "Inverter": 0, "Temp": 0, "Time": "10:29:05", "Uptime": 14454, "Vmod": 30.5, "Vopt": 52.0}, "173675E7": {"Date": "2019-10-12", "Eday": 0, "ID": "173675E7", "Imod": 2.2062500000000003, "Inverter": 0, "Temp": 0, "Time": "10:27:53", "Uptime": 14382, "Vmod": 31.375, "Vopt": 10.375}}}
dataLen:    0000
dataLenInv: ffff
sequence:   054d
source:     fffffffe
dest:       7415f893
function:   0080
/dev/ttyUSB0 <-- message: 3785 length: 22
data:       12 34 56 79 00 00 ff ff 4d 05 fe ff ff ff 93 f8
data:       15 74 80 00 7d 36
RS485 master ack timeout
dataLen:    0000
dataLenInv: ffff
sequence:   2f52
source:     fffffffe
dest:       00000001
function:   0302
/dev/ttyUSB0 <-- message: 3786 length: 22
data:       12 34 56 79 00 00 ff ff 52 2f fe ff ff ff 01 00
data:       00 00 02 03 97 2e
```

jbuehl, 2023-05-29:
After seeing your debug output I can see more than one problem.  The MASTER_GRANT (0x0302) message is sent by semonitor, the inverter responds with a POST_DATA (0x0500) message, and semonitor replies with an ACK (0x0080). The PROT_RESP_POLESTAR_MASTER_GRANT_ACK (0x039a) message is apparently never being sent by your inverter which results in the timeout message.  This is how the protocol worked in my 10 year old SE5000 inverter when this code was written, however it looks like SolarEdge may have changed it in your inverter.

Another problem is that even though 0x0500 messages are being received, validated, parsed, and acknowledged by semonitor, the inverter keeps sending the same ones over and over.  Messages with sequence numbers 0x054b and 0x054d appear more than once and contain the exact same data.  It may be that the inverter is expecting some other message as acknowledgement of receipt.

The only way I can think of to fix this would be to connect a device to your inverter that implements the current protocol and then monitor the messages on the RS485 bus.  I originally did this with two inverters configured as the master and slave, but that's not useful in this situation.

eriksl, 2023-05-29:
Yes, sadly I only have one inverter. Maybe at some time someone does.

On the other hand, I do get all of the data at some point, I now have successfully received data for most of the day today. I'll leave it running for a few more days and after that I'll probably have enough detailed information, the totals from the inverter (modbus) will probably be sufficient.

I have found the wrong date issue btw. Apparently my laptop had a wrong date in it's hardware clock, OS clock was incorrect after reboot, and somehow it managed to send it's date-time to the inverter, I am not sure if semonitor sents messages containing timestamps? Somehow at some point the inverter adapted this incorrect date-time, weird.

So I am actually quite content for the moment. Thanks for your effort!

jbuehl, 2023-05-29:
Yes the inverter does get its date from semonitor.

closed 2023-09-09 by eriksl