jbuehl/solaredge #182: semonitor mis-identifies the serial numbers on some of my optimizers

eggehad, 2023-09-09:
Thank you all for this wonderful software.  I was easily able to get up and running thanks to all of your hard work!

However I'm noticing an odd problem: semonitor.py seems to mis-identify the serial number(s) on some of my optimizers.  When I compare the output from semonitor.py (piped through seprint.py)  with my solaredge website data, I notice the serial numbers differ in the third digit, and only when the digit is zero or one!

...
Optimizers
  2023-09-08 19:17:02 100CC37B Eday:2387.25 Imod:   0.04  --->   serial number is actually 108CC37B
  2023-09-08 19:16:59 100DA754 Eday:2828.75 Imod:   0.00  --->   108DA754
  2023-09-08 19:16:30 100DA7CB Eday:2367.00 Imod:   0.16  --->   108DA7CB
  2023-09-08 19:19:55 100DA8BC Eday:2427.25 Imod:   0.00  --->  108DA8BC
  2023-09-08 19:17:49 100DC346 Eday:2434.75 Imod:   0.11  --->  108DC346
  2023-09-08 19:19:11 100E2594 Eday:2733.25 Imod:   0.01  --->  108E2594
  2023-09-08 19:20:22 1103072C Eday:1933.50 Imod:   0.04  --->  1183072C
  2023-09-08 19:18:23 110C657A Eday:   0.00 Imod:   0.02  --->  118C657A
  2023-09-08 19:17:34 111218FB Eday:1901.75 Imod:   0.08  --->  119218FB
  2023-09-08 19:19:38 123DB74B Eday:2012.25 Imod:   0.12  --->  is correct
  2023-09-08 19:19:21 124B5062 Eday:2082.25 Imod:   0.24  --->  is correct
...

Any idea what might be causing this?  
Could this be a parsing issue?

-david

eggehad, 2023-09-09:
I should have also mentioned, I'm noticing issues with the data output as well.  For example on Optimizer 118C657A, highlighted above, the Eday output shows up as 0, however,  SolarEdge shows that optimizer as having generated 2211 W*h today.   I put a current sensor on the panel, and there is indeed current there, which matches fairly well the current shown.   So zero is not the correct answer ;-)

jbuehl, 2023-09-09:
Turning off that bit in the serial number was done intentionally so the data would match the serial number in the labels on my devices when I wrote this code 10 years ago.  I don't remember which version of the optimizer serial numbers showed up on the SolarEdge website for me since I haven't connected to it in years.  I'm not sure if this applies to all types of optimizers or just the older ones that I have.  If you want to change how it behaves, look at the function `parseId` in data.py.

```
# remove the extra bit that is sometimes set in a device ID and upcase the letters
def parseId(seId):
    return ("%x" % (seId & 0xff7fffff)).upper()
```

eggehad, 2023-09-10:
Thank you!  100% solved now.  btw, I love your code!  Very nicely done :-)

closed 2023-09-10 by eggehad