jbuehl/solaredge #185: Exception while reading data

fozed1, 2023-09-10:
First and foremost - this is a great project
After several long days - I have got running for the most part.

My basic setup is:
- Inverter (192.168.2.249) --> eth1 --> Rasp3bian7 (192.168.2.249) --> eth0 --> router2 (Fritzbox)--> router1 (Hitron-ISP)--> Internet 
- The inverter is an SE4000H.

Trying to get data from the inverter, there is nothing coming up.
After that, I execute test1 and test2.
Test2 gives some info: "Exception while reading data: bytearray index out of range" (look at test2). 
Could be this "exception" the reason for nodata ??

Question
If yes how to solve this exception
Is there anybody who can give my some help ??
 
Thank you
[SE-testen.txt](https://github.com/jbuehl/solaredge/files/12568529/SE-testen.txt)

jbuehl, 2023-09-11:
It looks like your inverter is using encrypted communication to talk to the SolarEdge server which means that it isn't possible for semonitor to decode the data.  With newer inverters, the only way to access the data is to use the RS-485 interface.

fozed1, 2023-09-11:
Thank you for reply

fozed1, 2023-10-08:
The RS-485 interface installed
- Execute command and response (see below) ==> data not successfull.

Switched the data lines and it still doesn't work.
- Execute command (see below) == > no response at all.

My Solaredge settings are :
- RS485-1-conf >
- Device type > solaredge
- Protocol > slave
- Device id > 1

What I am doing wrong??
What is going wrong??
Is there anybody who can give my some help ??

[Solaredge_key.txt](https://github.com/jbuehl/solaredge/files/12840236/Solaredge_key.txt)