jbuehl/solaredge #186: ethernet active mode: failed to parse message

fredericve, 2023-09-12:
Hi,

I have an SE5000 inverter and would like to get some data about the optimizers in my setup.

I have my local dns servers pointing prod.solaredge.com and xprod.solaredge.com to a local server on my network and am running semonitor there while listening on ports 80, 22221 and 22222.

I first dumped the encryption key using sekey.py and passing that to semonitor.py now. So then I launch semonitor and getting this output:
```
./semonitor.py -t n -vvv -o output.json -k inverter.key -p 80,22221,22222
append: False
baudrate: 115200
commands: 
datasource: network
follow: False
interface: None
keyfile: <_io.BufferedReader name='inverter.key'>
logfile: stderr
master: False
outfile: output.json
ports: 80,22221,22222
record: None
slaves: 
type: n
updatefile: None
verbose: 3
xerror: False
opening network
waiting for connection on port 80
waiting for connection on port 22221
waiting for connection on port 22222
connection from 10.0.2.175:1766 to port 80
<socket> --> message: 2 length: 56
Saved rotating key length not correct. Not using
dataLen:    0022
dataLenInv: ffdd
sequence:   0351
source:     7f14f56f
dest:       fffffffe
function:   0503
--- Logging error ---
Traceback (most recent call last):
  File "/usr/lib/python3.9/logging/__init__.py", line 1079, in emit
    msg = self.format(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 923, in format
    return fmt.format(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 659, in format
    record.message = record.getMessage()
  File "/usr/lib/python3.9/logging/__init__.py", line 363, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Call stack:
  File "/root/git/solaredge/./semonitor.py", line 264, in <module>
    readData(args, mode, state, dataFile, recFile, outFile, keyStr)
  File "/root/git/solaredge/./semonitor.py", line 65, in readData
    processMsg(msg, args, mode, state, dataFile, recFile, outFile, keyStr, updateBuf)
  File "/root/git/solaredge/./semonitor.py", line 81, in processMsg
    (msgSeq, fromAddr, toAddr, function, data) = se.msg.parseMsg(msg, keyStr)
  File "/root/git/solaredge/se/msg.py", line 226, in parseMsg
    logger.data("Creating cipher object with key", keyStr)
  File "/root/git/solaredge/se/logutils.py", line 16, in _data_log
    self._log(LOG_LEVEL_DATA, msg, args, **kwargs)
Message: 'Creating cipher object with key'
Arguments: (b'2e5750737abb161f65792728f426a858',)
dataLen:    0000
dataLenInv: ffff
sequence:   0351
source:     fffffffe
dest:       7f14f56f
function:   0080
Encrypting message
Failed to parse message: can only concatenate list (not "bytes") to list
data:       22 00 dd ff 51 03 6f f5 14 7f fe ff ff ff 03 05
data:       c3 c2 e0 05 00 07 4b 8a fa dd b5 d6 bb ed c0 cc
data:       c8 ea f9 49 c1 08 46 0a b7 85 a0 7a 63 52 42 2a
data:       79 5b e7 9e
<socket> --> message: 3 length: 56
dataLen:    0022
dataLenInv: ffdd
sequence:   0351
source:     7f14f56f
dest:       fffffffe
function:   0503
--- Logging error ---
Traceback (most recent call last):
  File "/usr/lib/python3.9/logging/__init__.py", line 1079, in emit
    msg = self.format(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 923, in format
    return fmt.format(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 659, in format
    record.message = record.getMessage()
  File "/usr/lib/python3.9/logging/__init__.py", line 363, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Call stack:
  File "/root/git/solaredge/./semonitor.py", line 264, in <module>
    readData(args, mode, state, dataFile, recFile, outFile, keyStr)
  File "/root/git/solaredge/./semonitor.py", line 65, in readData
    processMsg(msg, args, mode, state, dataFile, recFile, outFile, keyStr, updateBuf)
  File "/root/git/solaredge/./semonitor.py", line 81, in processMsg
    (msgSeq, fromAddr, toAddr, function, data) = se.msg.parseMsg(msg, keyStr)
  File "/root/git/solaredge/se/msg.py", line 226, in parseMsg
    logger.data("Creating cipher object with key", keyStr)
  File "/root/git/solaredge/se/logutils.py", line 16, in _data_log
    self._log(LOG_LEVEL_DATA, msg, args, **kwargs)
Message: 'Creating cipher object with key'
Arguments: (b'REDACTED_DON_T_KNOW_IF_NEEDED',)
dataLen:    0000
dataLenInv: ffff
sequence:   0351
source:     fffffffe
dest:       7f14f56f
function:   0080
Encrypting message
Failed to parse message: can only concatenate list (not "bytes") to list
data:       22 00 dd ff 51 03 6f f5 14 7f fe ff ff ff 03 05
data:       c3 c2 e0 05 00 07 4b 8a fa dd b5 d6 bb ed c0 cc
data:       c8 ea f9 49 c1 08 46 0a b7 85 a0 7a 63 52 42 2a
data:       79 5b e7 9e
```
Anybody have a clue what I'm doing wrong?

fredericve, 2023-09-20:
Ah, I read in the updated readme that essentially the ethernet method no longer works, only RS485 will. But does RS485 include any data about the optimizers? That's really what I was looking for.

jbuehl, 2023-09-20:
It should, if you configure it to use the SolarEdge protocol.